damienmaguire/Stm32-vcu #34: Unintended behaviour of Throttle::CalcThrottle in Single Channel Mode

crasbe, 2022-11-17:
Hello,

I suspect that the ZombieVerter might have unexpected behavior in the Single Channel Mode. This is due to the following:

The function GetUserThrottleCommand reads the analog values of throttle1 and throttle2. This is done with the anain::Get function from the libopeninv. The configuration of the anain class is _not_ conditional on the use of one or two throttle inputs:
https://github.com/damienmaguire/Stm32-vcu/blob/342305287862c44ec5a7676cfab9e281694174d6/include/anain_prj.h#L10-L11

This means that in any case a value for throttle2 will be read from the GPIOs. If there's nothing connected, this must be treated as an undefined input. Probably it will be 0 because of the external pullup, but it doesn't have to be. One way or another, it should never be used in Single Channel Mode.

In the GetUserThrottleCommand function in line 80, the pot2val variable is set to Throttle::potmax[2] in Dual Channel mode, however it is not changed in Single Channel mode. This means that in Single Channel mode, the GPIO value of what was read from the GPIO (which shouldn't be used in the first place in Single Channel Mode) will be passed to CalcThrottle and most definitely be very different to Throttle::potmax[2]. 
In any case I'm not sure if line 80 would be a good idea anyways, since in Dual Channel mode, the pot2val variable isn't even the pot2val anymore.

utils::GetUserThrottleCommand: https://github.com/damienmaguire/Stm32-vcu/blob/ae9c2a6a5f20d4e57bed74c59c62a49651af16ca/src/utils.cpp#L49-L87

Now let's see what CalcThrottle does with the pot2val variable that's passed to it:

Throttle::CalcThrottle: https://github.com/damienmaguire/Stm32-vcu/blob/fe50dffdfaf16594421815492a9ef7a027f3e826/src/throttle.cpp#L97-L126

In line 102 it checks if potmin[1] is bigger than potmax[1]. This has three issues:
1) We shouldn't use potmin[1] and potmax[1] in Single Channel Mode, because they could be set to arbitrary values.
2) pot2val is set to potmax[1] in the previous function, so the expected behavior is, that the if condition is always true.
3) It is possible however, that potmax[1] is smaller than potmin[1], if the second potentiometer is inverted to the first one. This is checked in the Throttle::CheckDualThrottle function, but not in this function.

Regarding issue 2: scaledBrkMax is set to a constant value, since pot2val was set to a constant (potmax[1]) in GetUserThrottleCommand. But this only applies in Dual Channel mode!

This will subsequently have issues on the regen behaviour.

There is a small additional bug in line 76: in case of an error with throttle 2, the error message will say throttle 1 as well.

There is a huge flaw:
https://github.com/damienmaguire/Stm32-vcu/blob/fe50dffdfaf16594421815492a9ef7a027f3e826/src/throttle.cpp#L121

regenTravel is allowed to be set to 0, so this can be a division by zero, leading to +inf on the float!

https://github.com/damienmaguire/Stm32-vcu/blob/fe50dffdfaf16594421815492a9ef7a027f3e826/src/throttle.cpp#L100
One more thing: The variable "brknompedal" is not set anywhere. This means that it's probably set to 0 (default), but it's not specified.
This means scaledBrkMax will be set to -0.1f, since potnom is multiplied by zero:
https://github.com/damienmaguire/Stm32-vcu/blob/fe50dffdfaf16594421815492a9ef7a027f3e826/src/throttle.cpp#L102-L107
 

-------------------------

My proposed fixes are:
1) https://github.com/damienmaguire/Stm32-vcu/blob/7e766f7e7dde9166f7436f013650ff9150bd2905/include/param_prj.h#L307-L311
Change POTMODE_REGENADJ to POTMODE_SINGLECHANNEL.

2) Remove the pot2val = potmax[1]; from GetUserThrottleCommand. This can not be intended behaviour.

3) Change the identity of CalcThrottle to Throttle::CalcThrottle(potval, brake). Since the plausibility of pot2val has been checked, we don't need to send it to CalcThrottle (and we shouldn't, regarding issue 1 and 3).

4) Add a range check for regenTravel to avoid uninteded behaviour (division by zero).

5) Either discard brknompedal or set it to a value.

------------------------

I will prepare a pull request, but I can't test it at the moment.

crasbe, 2022-11-17:
Small thing: "throttleRamped" in throttle.cpp is not initialized before first use.

crasbe, 2022-11-17:
As with Commit https://github.com/jsphuebner/stm32-sine/commit/8dcec8613847427cead0093dae97aabf8458d528 in the OpenInverter code, the "- 1" needs to be deleted in Lines 58 and 63:
https://github.com/damienmaguire/Stm32-vcu/blob/master/src/throttle.cpp#L51-L71

Otherwise potval can become smaller than potmin, triggering the "potnom going to infinity" issue.

jsphuebner, 2022-11-17:
Yes though this has consequences further down:
https://github.com/damienmaguire/Stm32-vcu/blob/master/src/throttle.cpp#L102
It will disable regen even when the second pot is configured to be silent. I really recommend upgrading to the latest 
https://github.com/jsphuebner/stm32-sine/blob/master/src/throttle.cpp
where the behaviour has become more transparent (though to ultimate)

jsphuebner, 2022-11-17:
Also the rather large number of public member variables must be verified. They all default to 0 unless otherwise set. In the inverter ALL these variables are updated with values from parameters while the Zombie code does not explicitly set all of them

crasbe, 2022-11-17:
> Also the rather large number of public member variables must be verified. They all default to 0 unless otherwise set. In the inverter ALL these variables are updated with values from parameters while the Zombie code does not explicitly set all of them

Yes. In the OI code they are set in the main file from parameters, in the ZombieVerter code some are not set at all (like brknompedal). The Zombie doesn't even have the parameters.

https://github.com/jsphuebner/stm32-sine/blob/701875d26fdf3bfac641192990ff6174578368b6/src/stm32_sine.cpp#L302

jsphuebner, 2022-11-17:
Yes, it might not need all parameters but of course the variables must then be set to sane defaults. Might have been me who missed out on this, at least among others

crasbe, 2022-11-18:
With the pull requests #35 #36 #37 #38 this issue is fixed.

closed 2022-11-18 by crasbe