damienmaguire/Stm32-vcu #47: Add Soft IO to Vehicle Class

Mitchy6, 2023-01-13:
To allow more flexible installations on Canbus vehicles, allow ability for Vehicle Class to send status signals replacing the DIN_Start/Fwd/Rev etc.

This method, along with the dynamic pin assignment should allow more flexibility in using CAN signals.
For the purpose of the initial update, the vehicle signal can be OR'd similar to existing canio signal.

Reason for request:
Currently the OI inverter implementation requires DigIn signals, as they are simply mapped direct to the OI CAN interface.
Attempted to replicate similar E65 type canbus shifter, however that implementation seems to duplicate and complicate things more.
Simpler solution would be to allow Vehicle Class to Mimic DigIn signals to allow similar function.
Basically at the moment there is no way to control OI other than hardwiring signals into Zombie.

In the future it will likely allow us to over-assign the DigIO supplementing with CAN based signals. and repurposing existing IO that is unused due to signals being sourced on the CAN network.

*I am aware that CANio can be possibly configured for this purpose, but the main intent is to have zombie connect to OI, while keeping the Vehicle CAN separate.

jsphuebner, 2023-01-13:
The most recent vehicle classes on the master branch already support generating the start command and direction selection

Mitchy6, 2023-01-13:
Hi Johannes, Thanks for the reply.

I've been using those within the vehicle class.
I think in that case - the issue is that the OI inverter interface should be updated to align with the gear selection/Start() that the other inverters rely on. If we're standardizing on the Gear Selection, and higher level tag statuses (Start/Cruise), we can work that to update the OI inverter message.
At the moment the OI interface is looking at the core DigIO, so it never receives the correct indication if using the higher level tags.
For a quick test, I commented out the applicable Dig_In signals from Util.cpp, and write to the applicable 'Set Param' tag in my vehicle code. This works, but for a universal solution I've been trying to find a proper way forward.
Its easy enough to re-construct the OI inverter control message based on Param::dir & Param::T15Stat, Param::cruisestt and sending it out.
I hesitated as going the Dig_IO route seemed like the more fundamental place to adjust things.

Din_Brake will remain hardwired on both Zombie and OI...