damienmaguire/Stm32-vcu #49: Support configuration via CAN SDO

davefiddes, 2023-05-05:
I think it would be useful if the VCU allowed configuration via CAN. There have been several bug fixes for the CAN SDO protocol which prevented it from functioning. Also there have been improvements that allow CAN based tools to download the parameter and spot value database from the device and to accurately version this to prevent conflicts. 

Updating the version of [libopeninv](https://github.com/jsphuebner/libopeninv) to commit 73a8bb3631a8e207522dace05a7c246c598988e9 or newer will allow [openinverter-can-tool](https://pypi.org/project/openinverter-can-tool/) to configure, monitor and log data from the VCU.

damienmaguire, 2023-05-05:
Thanks Dave. I'll roll this in on the next release after 2.00 as I've just completed a huge batch of testing.

Tom-evnut, 2024-03-15:
Is this been verified to work? 

Johu posted on the forum about changing the ID to 0x603
https://openinverter.org/forum/viewtopic.php?p=68282&hilit=sdo#p68282 

I did a bit of playing on the bench with this, anyone else verified this works as it should so this can be closed?

jamiejones85, 2024-07-13:
I'm using it, so this can probably close.

closed 2025-01-25 by Tom-evnut