damienmaguire/Stm32-vcu #77: Simp BMS Charge current scale factor

Manny-electric, 2024-02-18:
I am using the simp BMS option in the VCU. The BMS_ChargeLim value in the web interface seems wrong. Looking in the simpbms.cpp, the chargeCurrentLimit  is divided by 1000. The simp_BMS datasheet states that this value is scaled by 10.

7yatna, 2024-02-18:
I have noticed the same issue and didnt bother to mess with it as i really dont need this feature. but glad that you found what the issue is.

Mitchy6, 2024-10-03:
@Tom-evnut 
You're likely the most active contributor. :) 
Just wondering if you can roll this into an upcoming release to purge it from the issue list.

Can confirm that the scaling is currently incorrect. I've corrected it in my offline build on the bench.
If not, I'll submit a PR whenever I get my build up to date with the list of recent fixes.

Tom-evnut, 2024-10-03:
@Mitchy6 Please provide corrective lines of code and I can push this in.

chrskly, 2025-01-22:
Submitted a PR that I think will fix this.

https://github.com/damienmaguire/Stm32-vcu/pull/146

Bear in mind I don't have a ZombieVerter, so, this is just based on me eyeballing the code.

Tom-evnut, 2025-01-25:
Code merged

closed 2025-01-25 by Tom-evnut