damienmaguire/Stm32-vcu #91: AC Charging with Leaf PDM not functioning because 0x1D4 is not being sent in charge mode

Romukeisari, 2024-04-29:
PDM goes to fail-safe, because 1D4 is not being sent in charge mode.

Add 1D4 to PDM code, so it gets sent when in charge mode. 
I have this working in my fork https://github.com/Romukeisari/Stm32-vcu-LeafLBC/blob/master/src/NissanPDM.cpp
Background: https://openinverter.org/forum/viewtopic.php?p=70232#p70232

damienmaguire, 2024-04-29:
Thanks. Will review.

Tom-evnut, 2024-05-01:
0x1D4 is sent for the inverter also, however in the STM32_vcu

This task is only called in a if function when the opmode is RUN  "402  selectedInverter->Task10Ms();"

Tom-evnut, 2024-05-01:
Create a "cleaner application" https://github.com/Tom-evnut/Stm32-vcu/tree/2.17A-Leaf-PDM-potential-fix

Romukeisari, 2024-05-07:
> Create a "cleaner application" https://github.com/Tom-evnut/Stm32-vcu/tree/2.17A-Leaf-PDM-potential-fix

If I may suggest, add a parameter read to check that also the Leaf inverter is there, otherwise this needs to be sent in all use cases so PDM doesn't go to fail-safe? Although fail-safe clears when powered off, wouldn't it be a preferred method of control to avoid the components going into fail-safe in all use cases?

Tom-evnut, 2024-05-07:
If I may suggest, add a parameter read to check that also the Leaf inverter is there

You would not use the PDM without the inverter really. Once testing is completed can review it and add it it in easily.

Tom-evnut, 2024-09-26:
@damienmaguire once you prove charging is working on your setup feel free to close this one.

Tom-evnut, 2024-12-02:
Release 2.20A fixes this, proven by @damienmaguire

Sector7e, 2024-12-27:
The update to 2.20A has fixed for my full 2013 Nissan Leaf stack with battery. I was unable to charge with version 2.17A.

Tom-evnut, 2025-01-05:
Only Gen3 PDM issues, will be its own topic.

closed 2025-01-05 by Tom-evnut