damienmaguire/Stm32-vcu #114: Positive throttle in reverse direction triggers regen brake light output

SomersetEV, 2024-09-28:
When in reverse direction, positive throttle above RegenBrakeLight value triggers the brake light function.

[params Regen light in rev.json](https://github.com/user-attachments/files/17176630/params.Regen.light.in.rev.json)

SomersetEV, 2024-09-28:
Line 536 in stm32_vcu.cpp turns torquePercent negative when in reverse. 
Line 550 sets BRAKELIGHT pin if torquePercent is below Param::RegenBrakeLight. In reverse torquePercent is negative, triggering the pin on acceleration instead of regen.

Tom-evnut, 2024-10-01:
https://github.com/damienmaguire/Stm32-vcu/pull/115 Pull request to fix, have not been able to test this on actual car.

Logic checks out in terms of the resulting signed value for comparison.

SomersetEV, 2024-10-30:
Testing with Outlander inverter firmware 2.20TM, no brake lights on reverse throttle but now brake lights in drive upon positive throttle. Parameters attached, from 2.17A but only Shunt parameter had to changed when updating to 2.20TM.

Reverse motor is active in this setup.

```
{
  "Inverter": 8,
  "Vehicle": 3,
  "Transmission": 0,
  "interface": 0,
  "chargemodes": 5,
  "InverterCan": 0,
  "VehicleCan": 0,
  "ShuntCan": 0,
  "LimCan": 1,
  "ChargerCan": 0,
  "BMSCan": 1,
  "OBD2Can": 1,
  "CanMapCan": 1,
  "DCDCCan": 1,
  "GearLvr": 0,
  "MotActive": 0,
  "potmin": 400,
  "potmax": 1950,
  "pot2min": 200,
  "pot2max": 970,
  "regenrpm": 1500,
  "regenendrpm": 100,
  "regenmax": -35,
  "regenBrake": -10,
  "regenramp": 1,
  "potmode": 1,
  "dirmode": 1,
  "reversemotor": 1,
  "throtramp": 10,
  "throtramprpm": 20000,
  "revlim": 9600,
  "bmslimhigh": 50,
  "bmslimlow": -1,
  "udcmin": 290,
  "udclim": 394,
  "idcmax": 5000,
  "idcmin": -5000,
  "tmphsmax": 85,
  "tmpmmax": 300,
  "throtmax": 100,
  "throtmin": -100,
  "throtmaxRev": 70,
  "throtdead": 10,
  "RegenBrakeLight": -10,
  "throtrpmfilt": 15,
  "Gear": 0,
  "OilPump": 50,
  "cruisestep": 200,
  "cruiseramp": 20,
  "regenlevel": 2,
  "udcsw": 320,
  "cruiselight": 0,
  "errlights": 0,
  "CAN3Speed": 0,
  "BattCap": 12,
  "Voltspnt": 390,
  "Pwrspnt": 3300,
  "IdcTerm": 0,
  "CCS_ICmd": 0,
  "CCS_ILim": 100,
  "CCS_SOCLim": 80,
  "SOCFC": 50,
  "Chgctrl": 0,
  "ChgAcVolt": 240,
  "ChgEff": 90,
  "DCdc_Type": 0,
  "DCSetPnt": 14,
  "BMS_Mode": 1,
  "BMS_Timeout": 10,
  "BMS_VminLimit": 3,
  "BMS_VmaxLimit": 4.12,
  "BMS_TminLimit": 0,
  "BMS_TmaxLimit": 50,
  "Heater": 0,
  "Control": 0,
  "HeatPwr": 0,
  "HeatPercnt": 0,
  "Set_Day": 0,
  "Set_Hour": 0,
  "Set_Min": 0,
  "Set_Sec": 0,
  "Chg_Hrs": 0,
  "Chg_Min": 0,
  "Chg_Dur": 0,
  "Pre_Hrs": 0,
  "Pre_Min": 0,
  "Pre_Dur": 0,
  "Out1Func": 4,
  "Out2Func": 6,
  "Out3Func": 9,
  "SL1Func": 8,
  "SL2Func": 0,
  "PWM1Func": 0,
  "PWM2Func": 0,
  "PWM3Func": 0,
  "GP12VInFunc": 0,
  "HVReqFunc": 0,
  "GPA1Func": 0,
  "GPA2Func": 0,
  "ppthresh": 2500,
  "BrkVacThresh": 2500,
  "BrkVacHyst": 2500,
  "IsaInit": 0,
  "Type": 1,
  "Tim3_Presc": 719,
  "Tim3_Period": 7200,
  "Tim3_1_OC": 3600,
  "Tim3_2_OC": 3600,
  "Tim3_3_OC": 3600
}
```

Tom-evnut, 2024-10-31:
Logic should not work motorrev. Needs verifying:
1.No motor reverse in forward yields brakelight output at right time
1.No motor reverse in reverse yields brakelight output at right time
1.Motor reverse active in forward yields brakelight output at right time
1.Motor reverse active in reverse yields brakelight output at right time

Tom-evnut, 2024-12-02:
Release 2.20A should fix this.