damienmaguire/Stm32-vcu #129: Open Inverter CAN control Clean up

Tom-evnut, 2024-12-03:
https://openinverter.org/forum/viewtopic.php?p=77330#p77330

1.Clean up CAN Counter
2.Clean up delayed start

Include other issue raised here: https://github.com/damienmaguire/Stm32-vcu/issues/117

Tom-evnut, 2024-12-14:
Also add mapping of OI Board CAN map to simplify configs

Tom-evnut, 2025-02-27:
Implemented https://github.com/damienmaguire/Stm32-vcu/tree/68432b9e2c54b648e5f4ae69d1ed47757c9356fa

Auto CAN Config works on bench. Waiting on bench test spins to verify controls and spot values then this can be closed.

jsphuebner, 2025-02-27:
Hi Tom,

I noticed you added SDO code to the class. Can you check whether the functions provided by cansdo.cpp can achieve the same and if not, how to tweak them to achieve what you want?
Here is an example where I use these functions: https://github.com/jsphuebner/FlyingAdcBms/blob/28bcee5f969dfb8eefe5a29063fb94b25628c799/src/bmsfsm.cpp#L121

Tom-evnut, 2025-02-27:
Hello Johannes,

Not quite sure what you mean? Are there ready made OI inter ECU CAN config functions available? It would be worth making nice clean ones. 

Rewriting it to SDO commands should be possible, I will have a look tomorrow. For CAN setup we will need 3 sub indexes per message. 

It is indeed something to work to getting "standardized", thanks for flagging this :)

jsphuebner, 2025-02-27:
Yes there are. I've only taken them far enough to work with the simple query in the BMS (and maybe slightly beyond). So now would be a good opportunity to flesh them out further

Tom-evnut, 2025-02-27:
Yes I already have a few ideas, I see you have the clearing of CAN maps in there which will be key. 

The largest issue still is all the upfront leg work of looking up Params for ECUs and building CAN map arrays. Which is not that painful really.

Main thing I will look at is a way of standardizing stepping through each CAN map line as its 3 SDO sends.

jsphuebner, 2025-02-27:
Yes I guess even that "trilogy" could be implemented in cansdo.cpp as it's the same every time. So like sdo->SDORemoteMap(paramid, canid, position....) and so on and it then handles all the handshaking

Tom-evnut, 2025-02-27:
Yup thats my idea, I have time tomorrow to dive into this. Would make it "easier" for others to replicate or tweak.

Tom-evnut, 2025-02-28:
Im lost I cannot get it to do it right. Can you please get the CAN_OI.cpp to be able to send simple sdo requests.

Probally need to pass the canSdo instance from STM32_VCU to it some how.

Tom-evnut, 2025-02-28:
Cannot work it out how to get the SDO to propagate can you please create an example? @jsphuebner

jsphuebner, 2025-02-28:
Will look into it this week

jsphuebner, 2025-03-03:
Try this one. I haven't tested yet.
CanMap::CANPOS map;
map.offsetBits = 8;
map.length = 16;
map.gain = 0.1f;
map.offset = 0;
` sdo->RemoteMap(nodeid, rx, id, &map)`

One cycle later you can call
sdo->SDOReadReply(&dummy)
It will return true if it worked (well, or sort of worked...)
Yes you will need to pass CanSdo to the class in question