damienmaguire/Stm32-vcu #145: TachoPPR not working with float value

Manny-electric, 2025-01-12:
currently the TachoPPR value is read as a INT

utils.cpp line 569 in void SpeedoSet(uint16_t speed)

```
float PulseGain = Param::GetInt(Param::TachoPPR);
```
should change to:

 `float PulseGain = Param::GetFloat(Param::TachoPPR);`

Tom-evnut, 2025-01-16:
Which vehicle needs a decimal pulse per revolution? Please add some info.

Most Speedos work off of ignition pulses so they would be multiples and not fractions.

Manny-electric, 2025-01-18:
I use this to drive the speedometer to show vehicle speed.  Think the code whas make to drive to tachometer, because of the 750 min and 7500 max.

But sins the variable is declared as a float and the web interface is a float. I assumed that was meant to be a float.

Tom-evnut, 2025-01-18:
Yea the web interfaces makes everything a float. I do not agree with doing that.

Yes it is to drive a tachometer. Could look at introducing an option to have it behave as a speedometer.

Proposal, new output type:
1. Start from 0
2. Float interoperation for frequency control

Sector7e, 2025-01-18:
Tom,
So the current 750 minimum is why the tach on my truck starts at 750, even though the motor isn't spinning yet. I was unable to find a way to set this to 0, so your proposal would fix my tach issue.

Tom-evnut, 2025-01-18:
Some tach signals are used to enable things like power steering ect. 

Will look at adding the "speedometer" output that allows you to do a tach that shows actual rpm.

Manny-electric, 2025-01-18:
A speedo type would be good.

I have it set to start form 5 km/h. To avoid the speedo jumping at low speed. I think the timer might be loaded with more than 16 bits. could not find a clear datasheet on the timer.

The tach starting from 750 seems good to me. Can act like drive enable indicator

Tom-evnut, 2025-02-12:
Gents pull request is made https://github.com/damienmaguire/Stm32-vcu/pull/158

I bench tested this, it only work on Vehicle set to Classic

Will close once other test